prusa3d/PrusaSlicer #10724: Cannot slice variable layer Hight with organic supports

valkires1, 2023-06-02:
### Description of the bug

I don't know if this is a bug or a limitation and this should be in the suggestions section but ill assume it's a bug.

Any object that is selected to have organic supports and variable layer Hight has a error that says "variable layer Hight is not supported with organic supports"

[image]

### Project file & How to reproduce

input file -> enable organic supports -> select variable layer Hight

[Printed Into Existence_Dragon Skull (2).zip](https://github.com/prusa3d/PrusaSlicer/files/11632070/Printed.Into.Existence_Dragon.Skull.2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta3+win64

### Operating system

Windows 10

### Printer model

i3mk3s+

rtyr, 2023-06-02:
Expected behavior, mentioned in release notes. Also a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9462.

closed 2023-06-02 by rtyr

neophyl, 2023-06-02:
This limitation is mentioned in the release notes of the software you downloaded and has already spawned about 20+ issues.