prusa3d/PrusaSlicer #10725: support tree

isaacbezerra3d, 2023-06-02:
### Description of the bug

quando fatiar a peça, aparecem buracos na construção do suporte.
Na versao alpha6 nao acontecia esse problema no suporte.

### Project file & How to reproduce

use peça que tenha suportes.
escolha suporte organico.
e depois de fatia aparece esses buracos no meio do suporte.
[bug-suport-organic-beta-3-260.zip](https://github.com/prusa3d/PrusaSlicer/files/11634993/bug-suport-organic-beta-3-260.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

windos 64 beta 3 versao 2.6.0

### Operating system

windos 10

### Printer model

ender v2, s1 pro,

bubnikv, 2023-06-07:
Description of the bug
when slicing the part, holes appear in the support construction.
In the alpha6 version this problem did not happen in support.

Project file & How to reproduce
use part that has supports.
choose organic support.
and after slice appears these holes in the middle of the support.

jd3dgiant, 2023-06-20:
I just ran into this issue switching from alpha 6 to rc2 of 2.6.0. Easier to get it to happen if the organic supports generated start with a really large base and support multiple parts of the print part from bottom to top versus a large organic base that only supports something at the top. Created a file with a scaled box and multiple scaled cylinders that cause it where as the same file in alpha6 is fine. Changing the Branch Diameter Angle from default 5 to 2 will move the issue to another organic support.
[image]
[image]
[ForceOrganicsError3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11794893/ForceOrganicsError3mf.zip)

jd3dgiant, 2023-06-22:
Wanted to bump this thread, as this is still happening in the 2.6.0 release version.

forest-hermit, 2023-09-05:
Keep on happening in the 2.6.1 RC2 version.

Swarook, 2024-05-28:
Still having this issue. I see that the most people do is go around telling other people that this thread exists but no one solved it. Lots of effort making sure no one posts this again but no one fixing it. 
Just go around telling everyone to stop making duplicate posts and you get to be a top contributor.

AlexStOd, 2024-09-26:
Are there any plans to fix the issue? More than one year have passed...

franktastical, 2024-10-10:
I am also running into this on PrusaSlicer-2.8.1+MacOS-arm64-202409181355, as seen here: https://imgur.com/a/n93qPzM

Riotpunchbeats, 2025-01-27:
still appear to be having this issue or a similar one on 2.9.0 on windows 11

[image]

rovili, 2025-04-14:
Is there an progress on this? Still happening on the newest version. 
Also supports on big branches start mid air.

SachCZ, 2025-04-14:
There is no progress on this.

rcmaniac25, 2026-09-13:
Still an issue on 3.0 alpha11

[image]

JimBim89, 2026-10-30:
Also run into this, would be nice if #15508 is looked at for next 3.0 alphas or even 2.9.X maintenance releases.