prusa3d/PrusaSlicer #10728: Typo in Print Settings - German translation

JWartenberg, 2023-06-03:
### Description of the bug

Typo in in Print Settings: "**Aü**ßere Perimeter zuerst drucken" instead of "**Äu**ßere Perimeter zuerst drucken"

Druckeinstellungen / Schichten und Perimeter / Erweitert

[image]

### Project file & How to reproduce

Start Print Settings

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 3

### Operating system

windows 11

### Printer model

MK3S+ with MMU2

YuSanka, 2023-06-05:
Thanks for your observations :) We will fix it for next release.

Jan-Soustruznik, 2023-06-05:
SPE-1735

lukasmatena, 2023-06-09:
Fixed in [2.6.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc1). Closing.

closed 2023-06-09 by lukasmatena