prusa3d/PrusaSlicer #10729: Prusaslicer 2.6.0-beta3 crashes when printing to physical printer (via octoprint)

TysonCross, 2023-06-03:
### Description of the bug

When sending Gcode to print using a physical printer (imported from previous versions of Prusasclicer), connected via octoprint API Key, Prusaslicer immediately crashes.

In previous Prusaslicer 2.6.0-alpha5+MacOS-x64, the gcode would be sent to the printer and the application would show a floating corner dialogue with a progress bar, or an error if the printer was not online and connected.

The printer passes the connection test via the edit/configure menu.

[image]

### Project file & How to reproduce

[test_crash_project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11642242/test_crash_project.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta3+MacOS-x64

### Operating system

Mac OS Mojave 10.14.6 (18G9323)

### Printer model

Prusa MK3S

sarusani, 2023-06-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10711

Try the new beta4 error should be fixed.

Jan-Soustruznik, 2023-06-05:
Hello, @TysonCross , 

this problem should be already fixed in latest beta4. Could you check, please? 

link beta4: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta4

Thank you, have a nice day

TysonCross, 2023-06-05:
I have confirmed that the issue has been fixed in beta4, thank you. (Please accept my apologies for the duplicate issue as well)