prusa3d/PrusaSlicer #10735: Ejecting of device has failed

brandonyoung4d, 2023-06-04:
### Description of the bug

When ejecting SD cards from PrusaSlicer, I get an error of "Ejecting of device has failed" every time. This does not happen with flash drives being ejected from PrusaSlicer.

### Project file & How to reproduce

This happens for all projects.
[SpoolRing.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11647338/SpoolRing.3mf.zip)

1. Slice model
2. Export gcode to sd card
3. Wait for export to finish
4. Click eject button
5. Error occurs

[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+MacOS-arm64

### Operating system

MacOS 13.3.1 (a) (22E772610a

### Printer model

MK3S+

Jan-Soustruznik, 2023-06-05:
Hello, @brandonyoung4d ,  Could you check with latest released beta4, please? link: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta4

thank you, have a nice day

brandonyoung4d, 2023-06-05:
Tried a flash drive as well with beta4, the flash drives ejects immediately like it should

Jan-Soustruznik, 2023-06-05:
Hello, @brandonyoung4d, I am little bit confused, problem is solved on your side?

brandonyoung4d, 2023-06-05:
The issue still persists, for SD cards even with beta4.

I made sure to be clear in the description about this as well. The issue is for SD cards and NOT flash drives.

I've also tested this on different Macs and have the same issue.

Jan-Soustruznik, 2023-06-06:
@brandonyoung4d , 

Did you try with different SD cards?

brandonyoung4d, 2023-06-14:
I just updated to PrusaSlicer 2.6.0-rc2 and the issue is now gone. Going to try a few more times, with different SD cards to make sure though

brandonyoung4d, 2023-06-19:
Sorry, my results have been inconsistent so it's hard to nail down what's going on. As of now 
* Issue occurs regardless of PrusaSlicer version
* Issue occurs only with a specific brand of SD card (I've tried multiple different SD cards from this brand though)
* Sometimes issue doesn't occur under repeated steps
* Issue doesn't occur if I eject the SD card using Finder on macOS

The fact that Finder can eject the disk every time makes me wonder if the API call PrusaSlicer is using is somehow failing to handle a situation where data is still being written to the SD card.

brandonyoung4d, 2026-09-02:
I can no longer reproduce this issue

closed 2026-09-02 by brandonyoung4d