prusa3d/PrusaSlicer #10736: PrusaSlicer 2.6 Beta4 snug support is difficult to remove

CrashCash, 2023-06-04:
### Description of the bug

I am using PrusaSlicer-2.6.0-beta4+linux-x64-GTK2-202306021356 on Linux (Debian Testing)

I am seeing issues with snug support. To demonstrate this, you need a tunnel about 20mm wide, 30mm long, and 12mm tall (about 30min print) - this is cut from a larger project so I had to add a brim.

In PrusaSlicer Beta3 and previous, you could print with snug supports and it would easily and cleanly snap off, leaving a clean smooth surface.

In Beta4, the support is extremely difficult to remove and leaves more of a scar. I have to use a sharp tool to pry it off the model.

I am using a 0.4mm nozzle with 1.75mm Hatchbox filament.

### Project file & How to reproduce

Print attached 3mf file and try to remove support.

Print Settings: 0.15mm QUALITY MK3
Printer: Original Prusa i3 MK3
Supports: Support on build plate only
Infill: doesn't matter
Support Style: Snug
Brim: yes
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/11647504/test.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta4+linux-x64-GTK2-202306021356

### Operating system

Devuan GNU/Linux daedalus/ceres (Debian w/o systemd)

### Printer model

Prusa i3 MK3 fw: 3.12.2-5713 0.4mm nozzle

rtyr, 2023-06-05:
The only difference I can see is bridging angle.

beta3/beta3/beta4

[image]

@Godrak

bubnikv, 2023-06-07:
Fixed with 02f69575ef580ef3dd8ed0ea7a82646157de5335

closed 2023-06-07 by bubnikv

CrashCash, 2023-06-10:
I just installed RC1 and it's a noticeable improvement. Thank you.