prusa3d/PrusaSlicer #10743: Adding negative Volume wont generate bridges when perimeters collide

ericgs, 2023-06-05:
### Description of the bug

After importing an object like a small cylinder and adding a negative volume, and place it on the base to create a partial hole, if there are many perimeters and the out perimeter collide with the new hole ones, instead of a bridge you get more perimeters on the layer above the hole
[image]
[image]

### Project file & How to reproduce

-Create a new project
-add a cylinder
-add a negative volume cylinder
-make it smaller
-place it at the middle of the original cylinder and lower it to be at the base
-slice the model with many perimeters so they can collide
-inspect first layer above the hole

[perimetersAndBridges.zip](https://github.com/prusa3d/PrusaSlicer/files/11654754/perimetersAndBridges.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 beta4

### Operating system

Windows10

### Printer model

Artillery Hornet

ericgs, 2023-06-07:
you are right, that solves the problem however i think there is another problem, i don't think this is the expected behavior, unless spanish translation is wrong, isn't this supposed to only apply where no bridge can be made?

Jan-Soustruznik, 2023-06-07:
@ericgs, I think the behavior is correct in your attached project. In your case, it is possible to use both ways - bridges or extra perimeters, but new feature can get better results(smooth surface, less supports, print stability) instead of the usual bridges, so the algorithm is also activated for this cases(if the function is enabled).

ericgs, 2023-06-07:
@Jan-Soustruznik , i understand that is a new feature, just dont understand why it gets enabled when there is no need for it, and only when the outer perimeters reach the hole in the middle. the documentation says "This functionality detects overhang areas where bridges cannot be anchored" and they clearly  can, but if you say this is the expected behavior i´ll take your word on it and just leave it off

Tupson444, 2026-09-18:
maybe still relevant