prusa3d/PrusaSlicer #10744: Bugreport slicing

girenloland, 2023-06-05:
### Description of the bug

Can not slice when selecting 4 perimeters with this particular settings. I can slice when choosing 1-2-3 walls, if i choose 4 perimeters I get error and Prusa crash to desktop.
See project file and screenshot
At the current project it is set to 3 walls. Change it to 4 walls, slice, then it crash "fatal error"
[image]
[Flexi Fox prusa crash.zip](https://github.com/prusa3d/PrusaSlicer/files/11655360/Flexi.Fox.prusa.crash.zip)

### Project file & How to reproduce

Just select 4 walls instead of 3. Then it crashes

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 3

### Operating system

Windows 11

### Printer model

Modified Ender 3 V2

lukasmatena, 2023-06-06:
Thanks. I reproduced this in current master and even in 2.6.0-alpha2, but not in 2.5.2.
The crash occurs in `Generator::getTreesForLayer` because `layer_id` index is out-of-range.
SPE-1753

Jan-Soustruznik, 2023-06-12:
Issue should be fix in latest release - https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc1

Closed as solved

closed 2023-06-12 by Jan-Soustruznik