prusa3d/PrusaSlicer #10746: drop box in filament settings to small

rafibar, 2023-06-06:
### Description of the bug

the drop box height is incorrect
[image]

### Project file & How to reproduce

[Untitled.zip](https://github.com/prusa3d/PrusaSlicer/files/11660930/Untitled.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta4+win64

### Operating system

win 11

### Printer model

mk3s+

Jan-Soustruznik, 2023-06-06:
SPE-1758

lukasmatena, 2023-06-09:
Thanks for the report. Fixed in [2.6.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc1). Closing.

closed 2023-06-09 by lukasmatena

rafibar, 2023-06-10:
I saw the correction in the new update rc1.
Thank you very much.

Max-Headroom, 2023-06-10:
Tested. Fixed. 
Thank you.

Win 10