prusa3d/PrusaSlicer #10749: Top z distance set to .4

KireanR, 2023-06-06:
### Description of the bug

Whatever I set my top z distance to the slicer ignores it and substitutes a .4 distance while making the layer above the support .4 as well. I tried this on a fresh install of the most recent stable version to the newest beta on a windows machine and a mac and all have the same bug for me. This causes horibble overhangs on models that you simply cannot get away from printing supports for.

I expected to be able to select a top z distance and have the slicer listen to that choice instead of deciding it's method is the best cause it's not
[image]
[image]
[image]
[image]

[image]

### Project file & How to reproduce

Activate supports, set a top z distance, check gcode
[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/11662993/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Happens with latest stable release up to the most recent beta on Windows and Mac

### Operating system

Windows 11 and Mac Os 13.3.1

### Printer model

Neptune 3 Plus

neophyl, 2023-06-06:
You need to attach a zipped up 3mf project file saved from PS, not the output gcode.  No one can check any settings or model placement , geo etc without it .  gcode is just not very useful in tracking down these issues.

KireanR, 2023-06-06:
That solved the issue,  you may want to set that to off by default in the next update. Could potentially throw off newbies like myself who are more thinking it's a setting I changed then it's a default setting that is causing this.