prusa3d/PrusaSlicer #10750: like wtf????????

Drlevinsk, 2023-06-06:
### Description of the bug

uploading perfectly fine 30 seconds prior then i suddenly get this.
Error uploading to print host:
HTTP 403: {"error": {"code": 403, "message": "Forbidden", "traceback": "Traceback (most recent call last):\n\n File \"/usr/share/moonraker/moonraker/components/file_manager/file_manager.py\", line 586, in _finish_gcode_upload\n can_start = await self._handle_operation_check(check_path)\n\n File \"/usr/share/moonraker/moonraker/components/file_manager/file_manager.py\", line 309, in _handle_operation_check\n raise self.server.error(\"File currently in use\", 403)\n\nutils.ServerError: File currently in use\n\n\nDuring handling of the above exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File \"/usr/share/moonraker/moonraker/app.py\", line 807, in post\n result = await self.file_manager.finalize_upload(form_args)\n\n File \"/usr/share/moonraker/moonraker/components/file_manager/file_manager.py\", line 451, in finalize_upload\n result = await self._finish_gcode_upload(upload_info)\n\n File \"/usr/share/moonraker/moonraker/components/file_manager/file_manager.py\", line 590, in _finish_gcode_upload\n \"File is loaded, upload not permitted\", 403)\n\nutils.ServerError: File is loaded, upload not permitted\n\n\nDuring handling of the above exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File \"/usr/share/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1704, in _execute\n result = await result\n\n File \"/usr/share/moonraker/moonraker/app.py\", line 810, in post\n e.status_code, str(e))\n\ntornado.web.HTTPError: HTTP 403: Forbidden (File is loaded, upload not permitted)\n"}}

### Project file & How to reproduce

just when i try to upload using octo.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

 2.5 

### Operating system

windows 8

### Printer model

ender 3v2, ender 6, ender 5 pro

plampix, 2023-06-06:
Looks like the file you're uploading is already loaded, and cannot be overwritten. Though PrusaSlicer is reporting the issue, it's clearly caused by moonraker (and probably for a good reason). PrusaSlicer is doing exactly what it should.

Drlevinsk, 2023-06-10:
all good forgot to close. fat fingers was the issue.lol

closed 2023-06-10 by Drlevinsk

Tomblarom, 2024-12-14:
Had the same issue. Turned out I used german umlauts, as the german word for "Body" is "Körper" in Fusion. This sucks.