prusa3d/PrusaSlicer #10754: Bridge Infill suddenly where it's not needed 2.6.0-alpha6 -> beta4

TylonHH, 2023-06-06:
### Description of the bug

As you can see in the screenshot, I sliced the same project with those versions. It's a simple solid cylinder sliced in vase mode with 3 bottom layer. The cylinder has a light radius on the bottom.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6 -> beta4

### Operating system

Win10 64bit

### Printer model

Snapmaker J1

Jan-Soustruznik, 2023-06-07:
Hello, @TylonHH , Could you share the project file(.3mf) where the problem occured, please?

TylonHH, 2023-06-07:
done. I updated the 1st comment. Just rename the .txt

TylonHH, 2023-06-11:
this issue doesnt exist anymore in 2.6.0-RC1

closed 2023-06-11 by TylonHH