prusa3d/PrusaSlicer #10757: Different support types (styles) on diffeent supports

valkires1, 2023-06-07:
Allow the option to print different types of supports per part. 

When printing some parts print better with the legacy supports where as others need the organic. There is the menu to change some of the settings but not the type.

[image]
[image]

neophyl, 2023-06-07:
Slightly different wording/details but pretty much the same functionality already requested in https://github.com/prusa3d/PrusaSlicer/issues/9785

Jan-Soustruznik, 2023-06-07:
Hello, @valkires1 , thank you for you feedback. This feature request already exist. I closed this ticket as duplicate. Have a nice day.

closed 2023-06-07 by Jan-Soustruznik