prusa3d/PrusaSlicer #10759: Lost Gcode icon on layer slider where colour change occurs

joantabb, 2023-06-07:
### Description of the bug

Good morning 
I am a moderator on the Prusa forum, 
About 6 months ago, a user reported issues with the Gcode marker on the layer height scroll bar where colour changes are added

Here is a link 
https://forum.prusa3d.com/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/prusaslicer-2-5-0-vertical-slider-doesnt-show-all-custom-gcode-entries/#post-629908
[image]
image linked above

this seems to be an issue from earlier slicer builds that persists in prusa slicer 2.6.0 Beta4

### Project file & How to reproduce

Load STL, 
Slice STL, 
In the sliced model window, choose a height for a colourchange, 
click Layer height scroll bar and choose colourchange

colour change occurs but the gcode marker that allows you to easily delete the colourchange does not appear. 
[Two colour vase Slicer 2.6.0 Beta 4.zip](https://github.com/prusa3d/PrusaSlicer/files/11675701/Two.colour.vase.Slicer.2.6.0.Beta.4.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PS2.5.2, PS 2.6.0 Beta 2 and Version 2.6.0-beta4+win64

### Operating system

Windows 11

### Printer model

Mk3S

sarusani, 2023-06-07:
It only appears when you select the layer. Maybe I'm wrong, but I think that's how it always worked.
 
[image]

neophyl, 2023-06-07:
Take a closer look at Joan's screencap and compare it to yours.  Yours has the icon visible, its grey with the x when on the colour change layer but at other points on the slider its orange with the + sign.  Even when its at the top.  

The images Joan posted are missing the icon completely.  
Which does beg the question if the icon isn't there at all how did the colour change get added ?  Or does it disappear for this user after they add the first one ?

When I load the project in I too have no issue with the icon not being on the slider, works perfectly for me and is always shown.

sarusani, 2023-06-07:
> Take a closer look at Joan's screencap and compare it to yours. Yours has the icon visible, its grey with the x when on the colour change layer but at other points on the slider its orange with the + sign. Even when its at the top.
> 
You're right, but I was referring to the problem he mentioned:
> colour change occurs but the gcode marker that allows you to easily delete the colourchange does not appear.
> 
In his screenshot the marker to delete the color change  should not be visible, because he's not on the layer with the color change.
He never mentioned that he's missing the marker to add, just the one to delete.

joantabb, 2023-06-07:
Oh Dear, I have had my Gender changed as well :-)

Joan

sarusani, 2023-06-07:
> Oh Dear, I have had my Gender changed as well :-) Joan

Wow, I wasn't aware that using the wrong pronoun actually changes someones gender! That's crazy 😄 

But I'm still not sure what exactly is the problem you want to get fixed, in the ticket you say the marker for deleting/editing the color change is missing, but in the linked post it's about the colors in the layer bar (it's only one color even when there are two separate colors in the preview). And neophyl says it's about the missing + before a color change is added. I'm confused...

joantabb, 2023-06-08:
I am confused too, because the OP had only  one colour change and two colour change (Gcode) tags visible in one image (The Layerheight bar shows a totally different colour)
and in the other image they had one colour change (Gcode Tag), but the layerheight bar showed the same colour top to bottom!

and the images I generated showed no colourchange tag but the layerheight bar did show two colours

Should there be tags as the original poster shows. or should there not be tags?

regards Joan
[image]

Jan-Soustruznik, 2023-06-23:
Hello, @joantabb ,

Thank you for you feedback. I checked your reported problem. There are two options 1. color change 2. custom g-code , pause 

1.
Token for edit for color change is displayed only, when you move to layer with color change is - when you click on right button on mouse, you can edit or delete the color change. I think that this behaviour wasnt changed and works also same in old PrusaSlicer versions .. 2.4.2/2.5.0/2.5.1/2.6.0 ...

2.
custom g-code, pause there is placed marker by token always - that you can see on the second screen above. Because in contrast to the color change, the bar does not change color.

Let me know if I helped you. Do you think that should be changed? If yes, how?

thank you, have a nice day

joantabb, 2023-06-23:
Thank you for your reply, 
On this image there is a colour change on the model, but no colour change  in the layer height  scroll bar, 
[image]

I would prefer there to be a definite indication of where the colour change occurs so that the code is easily accessible for modification
regards Joan

closed 2026-09-29 by prusaslicer-bot[bot]