prusa3d/PrusaSlicer #10760: Different speeds of the Outer Perimeter with disabled "auto cooling" feature

TheMick16, 2023-06-07:
### Description of the bug

[image]
[image]
[image]

I downloaded the MK4 Printable parts and realised a different speed on the outer perimeter WITH DISABLED "auto cooling" feature. Especially on shiny Materials like Prusament PETG Jet Black you can see the different speeds. Of course not a big Problem but nice to fix.

### Project file & How to reproduce

Download like me a Prusa MK4/3 X End stl file and slice it with the Slicer. Don't forget to change the preview from Feature type to Speed (mm/s). 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 beta 4

### Operating system

MacOS 14 Sonoma / MacOS 13 Ventura

### Printer model

Prusa i3 MK3S

neophyl, 2023-06-07:
Without an attached project file its impossible to tell for certain but aren't those different speeds likely to be caught by the small perimeters speed logic  and so not be related to cooling at all ?

The reason for asking for a project (3mf file) is that it will contain the settings you used for all the profiles as well as any modifiers.  Without the project the slice will be completely different.  

For example the same part with my profile slices like this for speed
[image]

Note no speed variation on the outer perimeters.

TheMick16, 2023-06-09:
Hi, thank you for your reply😁. I tested the file on different Profiles for the MK3S and on every stock Speed Profile (0,2 Speed & 0,15 Speed) is it reproducible. But with any other profile it is not noticeable. I also added a Project file (Compressed in ZIP). 

[x-end-motor.zip](https://github.com/prusa3d/PrusaSlicer/files/11709014/x-end-motor.zip)

Thank you for your time and have a nice Day😉
Best regards 
Mick

neophyl, 2023-06-09:
Originally I suspected it might be the slow down in layer time on the filament settings as that catches a lot of people out.  

Then after seeing your project with Dynamic speeds enabled thought it could be that but it isn't.  The dynamic speeds is causing differences in areas but the thing that's causing the ones you outlines is the small perimeter speed settings.  The lengths looks they are just being caught by that.  

You can see that by making your small perimeter speed the same as your external perimeter speed and slicing.  The difference goes away at that point.
[image]

TheMick16, 2023-06-09:
😳Crazy! I tried and actually solve this Problem. I will definitely save this little change. Thank you for your help😁. I want to ask you something. Are you a developer team member of the Prusa slicer or just a dude likes to help somebody😅?

neophyl, 2023-06-10:
Just a user of the software like you, although I have been using it for a long time. I also test software for a living at the moment 😀 So trying to work out what is going on is kind of second nature by now.

closed 2026-09-29 by prusaslicer-bot[bot]