prusa3d/PrusaSlicer #10762: Organic support generation never finishes on large overhang model.

VH-fixer, 2023-06-07:
### Description of the bug

When I enabled Organic supports,  the organic support generation on this model goes on for hours stuck at 70%.  I finally kill the slicer.    The normal grid supports only take a 3or 4 seconds.   I use the default settings for the organic supports. 

I tested this model for organic supports on 2.6.0 Alpha 6 I believe and it worked within a minute or two.   It saved many hours over the grid support.   I did not print it.  

### Project file & How to reproduce

Open the 3mf file.  
Only 1 filament is used.   MMU not involved. 
Run the slicer.   
It slices and lays out the pattern on the model.  
Then it hangs at 70% generating the organic supports. 
[j50_top_cutout-merged-buggy260beta4.zip](https://github.com/prusa3d/PrusaSlicer/files/11681852/j50_top_cutout-merged-buggy260beta4.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 4

### Operating system

Windows 11 22H2

### Printer model

MK3s+ with MMU2S

JohnOCFII, 2023-06-08:
Importing your project and slicing on my MacBook Pro (arm M1) works fine and does not hang for me with Beta 4.  Might be Windows specific.
[image]

murk-sy, 2023-06-08:
Can confirm, slices supports within ~30 seconds on alpha6 while beta4 seems to move nowhere.
Using Windows 10.

VH-fixer, 2023-06-09:
YEEHAA, getting it to repeat is 90% of the submission battle.   Keep up the good work.

VH-fixer, 2023-06-16:
I did not close this!!!!! It is a major bug in my opinion.   How did I close it?

VH-fixer, 2023-06-16:
Same thing still happens in RC2

foreachthing, 2023-06-16:
Win11, PS2.6rc2: it just sits there at 70%& of support generated.
When switching profile to mine and enable organic support, it only takes a few seconds to slice. Good luck hunting 😮 !

VH-fixer, 2023-06-24:
Tested in 2.6.0 release.  It works now.

foreachthing, 2023-06-24:
@VH-fixer, for me, it only works when I disable Raft first.

VH-fixer, 2023-06-24:
I had 3 layers of rafting.  I changed to 4 and it stopped working on the production release of 2.6.0.   This focuses where the problem is then.  4 or more layers of rafting and it hangs.

Jan-Soustruznik, 2023-07-10:
SPE-1809

bubnikv, 2023-08-29:
The issue was indeed connected to raft. I twill be fixed in PrusaSlicer 2.6.2 final.

closed 2023-08-29 by bubnikv