prusa3d/PrusaSlicer #10763: Organic supports don't contact with model despite 0 contact distance

FurrySphere, 2023-06-08:
### Description of the bug

Potential duplicate of an older one #9752
Version: 2.6.0 beta4

Issue:
[image]

Current workaround is to set top interface layers to zero:
[image]

### Project file & How to reproduce

[LeftRight_Keys_Head.zip](https://github.com/prusa3d/PrusaSlicer/files/11684338/LeftRight_Keys_Head.zip)
TC:
1. Set "Top contact Z distance" to 0
2. Set supports "Style" to "Organic"
3. Set "Top interface layers" to 5
ER: Support interface layers touch the model
AR: There is a gap between interface layers and the model
WA: Set "Top interface layers" to 0

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta4-win64

### Operating system

Windows 11

### Printer model

Vostok

StephS, 2023-07-18:
I noticed a similar issue where organic supports seemingly ignore the "Top contact Z distance" setting. You expect the distance to change with this setting, but the the Z distance appears to stay equal to the layer height.
This is on 2.6.0 Win64 release version.

neophyl, 2023-07-18:
This was mentioned in the release notes for organic supports.  

To quote the relevant section -
The Organic supports currently pose the following limitations:

For large models with a lot of supports, 16GB RAM minimum is recommended.
**Top / bottom vertical separation distances are rounded to multiples of layer height.**
Variable layer height is not supported.
Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.
Rafts with Organic supports are buggy.

closed 2026-09-29 by prusaslicer-bot[bot]