prusa3d/PrusaSlicer #10764: Not able to add new fillament profiles

werkstrom, 2023-06-08:
### Description of the bug

When adding new filament configs (using wizard) they do not show up in the drop down selection list
[image]

When reopening the wizard, the filaments have been added somewhere, because they are preselected.
[image]

 However they do not show up in the drop down.

I have restarted the app, still not working.

### Project file & How to reproduce

Doing this as my first step so app is just started.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Beta4

### Operating system

Windows 11

### Printer model

MK3S

werkstrom, 2023-06-08:
Removing the alpha .ini file and restarting Prusa Slicer presents me with this.
[image]

Selecting "Install" something ocupies the computer for a bit. Then I am presented with the filament wizard again. Again, there are preselcted filaments, clicking finish the filament dropdown flickers like it's being reloaded/refreshed but still nothing to select in it. 
[image]

rtyr, 2023-06-08:
Could you please archive your whole configuration folder to the zip file and attach it here?

This option will point you to the configuration folder.
[image]

werkstrom, 2023-06-08:
Thanks for the quick reply @rtyr. Zip Attached
[PrusaSlicer-beta.zip](https://github.com/prusa3d/PrusaSlicer/files/11685318/PrusaSlicer-beta.zip)

werkstrom, 2023-06-08:
Tried importing config budle from 2.5.2. I believe I did that in the alpha I tried out some time ago.
[image]

However, Only one shows up and has some sort of red flag on it. (That I'm not sure why it has, but I'm sure you do 🤪)
[image]

werkstrom, 2023-06-08:
Well, well, well... It appears In the Beta, it appears the only set printer in the wizard (that should have been imported i guess, I haven't, to my knowledge set it up manually, but could be wrong). was an MK4, I have an MK3S. Changing that appears to have fixed the issue.
[image]

Not sure what is expected here, so I'll leave it to you to decide on closing the ticket or persuing a fix. 😁

rtyr, 2023-06-08:
Yes, new installation will enable only first printer by default.

I would say it is expected behavior, so I am closing this.

closed 2023-06-08 by rtyr