prusa3d/PrusaSlicer #10766: Layer conflict with organic support while using multiple copies and automatic part distribution

Sapheus, 2023-06-08:
### Description of the bug

Hey just made multiple copies of a small part. Normally there is an automatic placement oft the parts. But this time the slicer created a Layer hight conflict. the Parts are too close together because the organic support creates a larger first layer wich results in overlay of the first layer. The solution is to increase the distance between the parts. May be it is necessary to recalculate the distance of the copied and automatically placed Parts depending on the place that is needed to avoid overlapping of the support layer . 
Hope I could help a little to ameliorate your slicer ( wich I love ). Thank you for your work . 
Wonderful greetings from Saarbrücken - Germany.  

### Project file & How to reproduce

[3 Pictures.zip](https://github.com/prusa3d/PrusaSlicer/files/11691410/3.Pictures.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 beta 4

### Operating system

Mac OS Ventura 13.4

### Printer model

MK3s upgraded (no Printer Error)

Mundball, 2023-06-14:
Hello Prusa Dev Team,

this "bug" affects all types of support and brim too.

Is there a way to set/change the minimal distance between the objects placed on the bed with automatic placement?

lukasmatena, 2023-06-14:
@Sapheus Thanks for the feedback. This is a long existing issue, which was only made a bit more visible with the organic supports. The 2.6.0-beta4 and later should at least warn about the toolpath collisions after slicing.

@Mundball Yes. Right-click the Arrange icon in the top toolbar.

Mundball, 2023-06-15:
@lukasmatena 
Thank you.

closed 2023-08-29 by kubispe1