prusa3d/PrusaSlicer #10770: What is causing these calculation errors?

BakeJake, 2023-06-09:
### Description of the bug

Why does it take two minutes more to travel twice as fast, am I missing something?
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[3dbenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/11702445/3dbenchy.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta4

### Operating system

Windows 11

### Printer model

Creality Ender-3 S1 On Sonic Pad Moderately Modified.

Wolfmark, 2023-06-09:
Wipe code looks different.

For travel speed 250mm/s it looks like this:
```
;WIPE_START
G1 F12000
G1 X114.455 Y114.845 E1.72086
G1 X114.954 Y115.258 E1.58241
G1 X115.315 Y115.709 E1.45893
G1 X115.368 Y115.805 E1.4354
;WIPE_END
```

For travel speed 500mm/s it looks like this:
```
;WIPE_START
G1 F24000
G1 X114.455 Y114.845 E1.77813
G1 X114.954 Y115.258 E1.7089
G1 X115.315 Y115.709 E1.64716
G1 X115.588 Y116.2 E1.58712
G1 X115.701 Y116.527 E1.55014
G1 X115.832 Y117.038 E1.49376
G1 X115.871 Y117.549 E1.43899
G1 X115.868 Y117.582 E1.4354
;WIPE_END
```

So there are more movements when the travel speed is increased.
The code seems to be here: https://github.com/prusa3d/PrusaSlicer/blob/47cdd7d9c42e93d753e1ca9e5c01098b01dde5b2/src/libslic3r/GCode.cpp#L149

From what I see wipe_speed is set to travel_speed * 0.8.
And then the amount extracted/retracted per movement is 0.95 * extruder.retract_speed() / wipe_speed.

So an increased travel speed => increased wipe speed => reduced extracted/retracted amount per movement => more movements.

Note that I'm not a contributor and is the first time I saw the code, so I may be wrong.

Wolfmark, 2023-06-09:
From what I see in the code, there are 3 settings during wiping:

1. The travel speed, which in your case is 250mm/s or 500mm/s. The wipe speed is 80% of this value.
2. Retraction length, which in your case is 0.4mm.
3. Retraction speed, which in your case is 45mm/s.

When you increase the travel speed, you also increase the speed used during wiping. But the retraction length and retraction speed are the same. So yes, it can move faster, but must retract same length using same 45mm/s speed. In other words, you gain nothing.

The slicer generates more movements for a bigger difference between travel speed and retraction speed. For example:

1. when travel speed = 250mm/s and retraction speed = 45mm/s => 4 movements
2. when travel speed = 250mm/s and retraction speed = 250mm/s => only one movement
3. when travel speed = 250mm/s and retraction speed = 125mm/s => 2 movements

Have no idea if this is OK or can be improved. Is how the current algorithm works. But the number of movements can explain the difference in estimated time (which is an estimation, so maybe in reality there's no significant difference). I assume the printer doesn't spend too much time moving the extruder from one location to another without extruding/retracting, so increasing traveling speed alone doesn't actually reduce printing time. May help with stringing or other issues though...

BakeJake, 2023-06-09:
> From what I see in the code, there are 3 settings used during wiping:
> 
> 1. The travel speed, which in your case is 250mm/s or 500mm/s. The wipe speed is 80% of this value.
> 2. Retraction length, which in your case is 0.4mm.
> 3. Retraction speed, which in your case is 45mm/s.
> 
> When you increase the travel speed, you also increase the speed used during wiping. But the retraction length and retraction speed are the same. So yes, it can move faster, but must retract same length using same 45mm/s speed. In other words, you gain nothing.
> 
> The slicer generates more movements for a bigger difference between travel speed and retraction speed. For example:
> 
> 1. when travel speed = 250mm/s and retraction speed = 45mm/s => 4 movements
> 2. when travel speed = 250mm/s and retraction speed = 250mm/s => only one movement
> 3. when travel speed = 250mm/s and retraction speed = 125mm/s => 2 movements
> 
> Have no idea if this is OK or can be improved. Is how the current algorithm works. But the number of movements can explain the difference in estimated time (which is an estimation, so maybe in reality there's no significant difference). I assume the printer doesn't spend too much time moving the extruder from one location to another without extruding/retracting, so increasing traveling speed alone doesn't actually reduce printing time. May help with stringing or other issues though...

So, disabling that would be my best option if, I were going for speed? The number was just theoretical, I probably wouldn't use 500mm/s and do I understand you correctly? its making up for the loss in retraction by making more retractions?

Wolfmark, 2023-06-09:
Disabling what? Retractions? If so, yes, the print will be much faster ... and the result worse.

And no, there's no loss in retraction. Just that it uses more steps to retract the same 0.4mm, using the same 45mm/s speed. It doesn't retract more or less, slower or faster. It generates more steps and probably in this case doesn't matter.

BakeJake, 2023-06-09:
> Disabling what? Retractions? If so, yes, the print will be much faster ... and the result worse.
>
> And no, there's no loss in retraction. Just that it uses more steps to retract the same 0.4mm, using [...]

I mean if I disable the wipe before retract?

Wolfmark, 2023-06-09:
From what I see it estimates 3 minutes less without wiping. But probably stringing will be an issue. Note that I'm not that advanced in 3d printing, so I can't help you with this.

BakeJake, 2023-06-09:
> From what I see it estimates 3 minutes less without wiping. But probably stringing will be an issue. Note that I'm not that advanced in 3d printing, so I can't help you with this.

I'll have to try it out, I don't think stringing was an issue as I have my machine precisely tuned, I retract mostly at 0.4mm and 45mm/s. It used to be disabled and think it just made the seam a little more noticeable but maybe retuning retraction after turning off would fix those issues.

neophyl, 2023-06-10:
No. Moment on the whole wipe/retraction discussion as I don’t know enough about the code but I took a look at your project and the thing that’s slowing it down is your minimum layer time values on your filament cooling section. 

Change your slow down if layer time is below xx setting to a very small number and your actual print times will be reduced.   It’s common for fast printers to get hit with that.

BakeJake, 2023-06-10:
> No. Moment on the whole wipe/retraction discussion as I don’t know enough about the code but I took a look at your project and the thing that’s slowing it down is your minimum layer time values on [...]

For the most part I disable that on fast prints, wasn't aware I had it enabled! Good catch! I'll have to see if it persist with changing that. Still weird how it takes longer for a faster speed, in theory it should just be faster (but as explained above its a little more involved) I think it indeed needs a rewrite, or an option to change the percentage of speed!

closed 2026-09-29 by prusaslicer-bot[bot]