prusa3d/PrusaSlicer #10773: When changing height range modifier's number of perimeters, solid infill doesn't fill entire layer

Matszwe02, 2023-06-09:
### Description of the bug

Default 3 perimeters:
[image]

Height range modifier to 2 perimeters:
[image]

And 1 perimeter:
[image]

Arachne works with other perimeters, but it leaves gaps where the missing perimeters were.

### Project file & How to reproduce

Effect was observed on this file:
[[a]_Guidler_x1.zip](https://github.com/prusa3d/PrusaSlicer/files/11708085/a._Guidler_x1.zip)

- set height range modifier that includes top layer
- change number of perimeters (in this case to 1)
- see that solid infill doesn't fill exactly from wall to wall

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta3

### Operating system

Windows 10

### Printer model

Voron 0

neophyl, 2023-06-09:
The first picture shows infill, its printed in a rectilinear fashion and the angle its printed at allows it to go right up to the perimeters.  It will do that with rectilinear infill for 1,2 or 3 perimeters.

Your second 2 perimeters show TOPFILL (as its red not purple).  You have selected monotonic lines.  With topfill it cant quite fill that area even with 3 perimeters, for example 
[image]

When you select 2 or 1 perimeter its way more obvious of course.  There just isnt 'quite' enough room to get that extra line in.  You can adjust the settings accordingly to work around this.  
You could adjust the fill angle, for example 45 degrees
[image]
but even that will have small gaps where it cant fit in a line, in the corner for example.

Matszwe02, 2023-06-09:
> The first picture shows infill, its printed in a rectilinear fashion and the angle its printed at allows it to go right up to the perimeters. It will do that with rectilinear infill for 1,2 or 3 [...]

Good point. I saw the same effect without any changes to perimeter count, but that change highlights these gaps. I think arachne should take care of that tiny gap and make every line 0.01mm wider or only the last one. Also these empty "triangles" are frustrating as well that they appear everywhere

closed 2026-09-29 by prusaslicer-bot[bot]