prusa3d/PrusaSlicer #10776: Merged object delete error

viper0078, 2023-06-10:
### Description of the bug

[image]
Please try deleting only one object from the merged objects.

[image]
The name of the remaining object is not being displayed.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 beta4

### Operating system

win11 64bit

### Printer model

sm2

Jan-Soustruznik, 2023-06-12:
Hello, @viper0078 , 

Thank you for your feedback. That is correct behaviour. When you merge some models, then is the general model is named "Merged" - that is global name containts parts - merged models, when the parts deleted and only one parts is leave, then is not necessaty to dispaly any parts. So is visible only the general name. 

Or could you explayn how it should works? 

Thank you, have a nice day

viper0078, 2023-06-14:
OK. Since it seems to be a matter of personal preference, I will close it now.

closed 2023-06-14 by viper0078