prusa3d/PrusaSlicer #10777: error: "Access violation"

valkires1, 2023-06-10:
### Description of the bug

Had a object. enlarged it by 200% (sliced fine at normal size) and hit that error.

[image]

Original model: https://www.printables.com/model/249411-dragon-skull

Then if you simplify object again you get this message

[image]

### Project file & How to reproduce

Don't know what I did to cause it or why it happened.

[Uploading blackjack quality support test small.zip…]

(Had to simply object because original was 35mb zipped. error seems to still occur regardless)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0beta4

### Operating system

windows 10

### Printer model

mk3s+

valkires1, 2023-06-10:
[blackjack quality support test small.zip](https://github.com/prusa3d/PrusaSlicer/files/11711901/blackjack.quality.support.test.small.zip)
Think I didn't wait for the upload to finish so here's the file again

valkires1, 2023-06-10:
Note: I just installed 2.6.0rc1 and the problem doesn't occur. It may have been patched in the update?

Jan-Soustruznik, 2023-06-12:
Hello, @valkires1 , 

thank you for you info. Yes we had a several issues in beta releases with "Access violation". It seems that crash was caused by already fixed part of code.

I closed issue as fixed, have a nice day.

closed 2023-06-12 by Jan-Soustruznik