prusa3d/PrusaSlicer #10779: No Tool macros in G-codfe on slicing

Rob65, 2023-06-10:
### Description of the bug

On a printer with multiple extruders, no T command is generated if extruder 1 is the only extruder used.

If any extruder is the only extruder being used, the generated G-code does contain a T1, T2 (or other extruder).
But as soon as the first extruder is the only extruder being used, the T0 command is omitted from the G-code, as well as the T0 parameter on the M104 and M109 commands.

To reproduce this, create a printer with reprap G-code flavor, create 2 extruders and generate G-code for an object using just extruder 2. That file will contain T1 commands. Now change the extruder used to extruder 1, no T0 commands are in the file.

This results the current extruder being used and on a printer with a tool changer that results (in my case) in print failure since all extruders are parked by default (current tool T-1). Selecting T0 by hand before starting the print is not an option since that means the wipe and priming being done as part of the tool change is done at the wrong time.

### Project file & How to reproduce

[missing_tool_info.zip](https://github.com/prusa3d/PrusaSlicer/files/11712302/missing_tool_info.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-rc1+win64

### Operating system

Windows 10

### Printer model

Jubilee with Duet2 RRF 3.4.5

StevenCregan, 2023-07-01:
Current workaround seems to be using "Custom G-code" settings under the Printer Settings profile:
Here's an example from my toolchanger running reprapfirmware 3:

M104 ; Prevent Pusaslicer from adding temperature commands
; Run custom start G-code
T-1  ; Clear tool selection
G29 S1   ; Activate Mesh Bed Compensation and Load Saved Heightmap
; Set extruder temps based on filament profiles
{if temperature[0] > 30}
M568 P0 S{temperature[0]} R{temperature[0] - 30} ; set T0 active and idle temps
{else}
M568 P0 S{temperature[0]} R{temperature[0]} ; set T0 without delta
{endif}
{if temperature[1] > 30}
M568 P1 S{temperature[1]} R{temperature[1] - 30} ; set T1 active and idle temps
{else}
M568 P1 S{temperature[1]} R{temperature[1]} ; set T1 without delta
{endif}
{if temperature[2] > 30}
M568 P2 S{temperature[2]} R{temperature[2] - 30} ; set T2 active and idle temps
{else}
M568 P2 S{temperature[2]} R{temperature[2]} ; set T2 without delta
{endif}
{if temperature[3] > 30}
M568 P3 S{temperature[3]} R{temperature[3] - 30} ; set T0 active and idle temps
{else}
M568 P3 S{temperature[3]} R{temperature[3]} ; set T0 without delta
{endif}
T-1 ; Clear tool selection
T[current_extruder]  ; Pick up current tool
M116 P[current_extruder]  ; wait for extruder to reach temp
M140 S[bed_temperature] R0 ; set bed temp
M190 S[bed_temperature] ; wait for bed temp
M144 S1 ; Set bed as active

Rob65, 2023-07-02:
Using the T[current_extruder] seems to work (thanks!) so this is a good workaround
But it would still be good to fix this issue

closed 2026-09-29 by prusaslicer-bot[bot]