prusa3d/PrusaSlicer #10780: [Bug/regression]: New version is not checked/announced on startup

RichiH, 2023-06-10:
### Description of the bug

I started Prusa Slicer 2.6.0-alpha6 on 2023-06-10. I would have expected a notification on startup that 2.6.0-rc1 is available.

I am near certain that this worked reliably in the in the 2.4 release cycle. I believe it didn't work in the 2.5 one, but I am not certain. As such, I am marking it as a regression.

### Project file & How to reproduce

Start Prusa Slicer 2.6 alpha6

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

 2.6 alpha6

### Operating system

Linux

### Printer model

Mini (waiting for my multi-head XL....)

neophyl, 2023-06-10:
I just tried opening 2.6 alpha 5 and I also tried opening 2.6 alpha 6 and both times I was informed there was a prerelease 2.6.0-rc1 available.  
I also tried opening 2.6 beta 4 and was also informed there was a 2.6.0-rc1 available.

I then tried opening 2.5.1, in that instance I was informed about 2.5.2 being available.

That is everything I would expect.  I would not expect a full release version to inform me that any alpha/beat or rc was available.  I would expect that alphas/beats etc would inform that there was newer versions of non released software.  Which is exactly what I got.

This was on a Windows 10 computer.  Unless they have changed something server side in the last hour since you tried then there doesn't look to be an issue to me.

RichiH, 2023-06-10:
Oooh, it's in the bottom right; it used to be a modal. Sorry for the noise!

closed 2023-06-10 by RichiH