prusa3d/PrusaSlicer #10781: Multi extruder: insufficient priming when using "Prime all printing extruders"

viesturz, 2023-06-10:
### Description of the bug

When tools are changed, prusa retracts the filament of the previous tool by 10mm.
So at the end of a print, all but the last active tool are retracted by that amount.

When starting a new print, the initial priming process only extrudes arond 11mm which, combined with some oozing is not sufficient to prime the nozzle.

This,  together with https://github.com/prusa3d/PrusaSlicer/issues/10586 creates a really unrealiable wipe tower.

### Project file & How to reproduce

[cali_2-R.zip](https://github.com/prusa3d/PrusaSlicer/files/11713109/cali_2-R.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta

### Operating system

Win11

### Printer model

voron 2.4 with Revo

viesturz, 2023-06-10:
See the initial gcode below.

It seems to go like this:

- Start: T0 - 0 retraction, T1 - 10mm retraction from previous docking
- Change to T1
- Prime T1 with ~11m
- Change to T0
- Prime T0 with ~7mm

So this results in T0 primed with 7mm, but T1 only ~1mm.

```
M73 P0 R49
;TYPE:Custom
PRINT_START TOOL_TEMP=210 BED_TEMP=70
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G1 Z.15 F18000
;HEIGHT:0.3
;TYPE:Wipe tower
;WIDTH:0.5
;--------------------
; CP PRIMING START
;--------------------
M220 S100
G1 X7.000 Y0.260 F7200
M104 S195 T1 ; set new tool temperature so it can start heating while changing
T1
M109 S210 T1 ; set temperature and wait for it to be reached

G1 F18000
G1 X7.000 Y0.260
G4 S0
; CP TOOLCHANGE WIPE
G1 X67.000 E3.2600 F1584
G1 X66.250 Y0.760 E0.0490
G1 X7.000 E3.2192 F1800
G1 X7.750 Y1.260 E0.0490
G1 X66.250 E3.1785 F2198
G1 X7.750 F7200
; CP TOOLCHANGE UNLOAD
;WIDTH:0.5
SET_PRESSURE_ADVANCE ADVANCE=0
;WIDTH:0.5
G4 S0
G1 X67.000 Y0.260

;HEIGHT:0.3
;TYPE:Wipe tower
;WIDTH:0.5
G1 E-5 F2400
;WIPE_START
G1 F14400;_WIPE
G1 X90.012 Y21.886 E-5
;WIPE_END
M104 S205 T1 ; set temperature ;cooldown
M104 S195 T0 ; set new tool temperature so it can start heating while changing
T0
M109 S210 T0 ; set temperature and wait for it to be reached

G1 X67.000 Y0.260 F18000
G1 X67.000 Y0.260
G4 S0
; CP TOOLCHANGE WIPE
G1 X127.000 E3.2600 F1584
G1 X126.250 Y0.760 E0.0490
G1 X67.000 E3.2192 F1800
G1 X126.250 F7200
G1 F18000
G4 S0
G92 E0
; CP PRIMING END
;------------------
```

jettoblack, 2023-09-04:
I'm also having this issue.

closed 2026-09-29 by prusaslicer-bot[bot]