prusa3d/PrusaSlicer #10783: Temporary .tmp0 files cannot be read, deleted or otherwise managed by Windows

RalphBacon, 2023-06-11:
### Description of the bug

Sometimes PrusaSlicer leaves .tmp0 files in the gcode directory, to which neither Windows system nor any other admin has access to. The only way these are managed is by a complete system reboot (when they seem to disappear).

I request that full permissions are given to Windows Admin Group users to these files when created so that erroneous leftover files can be dealt with like any file, should the need arise.

### Project file & How to reproduce

Unknown how to reproduce, happens periodically. Screenshot of left over file and permissions shown:

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-rc1+win64

### Operating system

Windows 11

### Printer model

Prusa Mini+

HeyThatsMyToast, 2024-12-08:
Just got one of these as well, seems to just be a prusaslicer problem. You can probably use some tool to forcibly delete them. I'll do some testing.

RalphBacon, 2026-09-02:
So no "We investigated but could not reproduce", just a "We didn't investigate and can't be bothered now this is so old, and this is a great opportunity to close another bug report down". 

Again.

closed 2026-09-02 by RalphBacon