prusa3d/PrusaSlicer #10784: Allow advanced mode entering retraction settings in case of Firmware Retraction (currently disabled)

Henk72, 2023-06-11:
**Is your feature request related to a problem? Please describe.**
If Firmware Retraction is enabled, it is no longer possible to set retraction settings. 

**Describe the solution you'd like**
Allow the user in advanced mode to enter retraction settings if Firmware Retraction is on.

**Describe how it would work**
This allows a user to set all retraction settings in the slicer, depending on different filaments etc. The advanced user could manually pass these parameters to the printer firmware in the custom G-code settings for START_PRINT for example for Klipper.
The big advantage is that all retraction settings are slicer controlled, but can be overruled during printing. 

**Describe alternatives you've considered**
Manually adding the parameters in the START_PRINT custom G-code.

**Additional context**
Best of both world for advanced users.

bodja, 2025-03-10:
It is actually possible. See https://help.prusa3d.com/article/list-of-placeholders_205643 it is described how to get a variable name (see screenshot the marked it red).

For retraction settings go to the filament overrides and hover over e.g. "Retraction length", it will display the name of the variable (`filament_retract_length`). So here you go: `SET_RETRACTION RETRACT_LENGTH={filament_retract_length[0]}`

Henk72, 2025-03-13:
Thank you for the pointer. This is a good workaround. By editing the condig start_filmament_gcode I can put your suggestion in there and make it available to all filaments.

I have closed the issue.

closed 2025-03-13 by Henk72