prusa3d/PrusaSlicer #10786: 2.6.0-rc1: Dynamic overhang speed settings do not apply in modifiers

pdcook, 2023-06-11:
### Description of the bug

Dynamic overhang speed settings (introduced in 2.6.0-alpha2) work well when set from the global print settings. However, adding a modifier which contains settings for these dynamic overhang speeds does not seem to apply those speeds at all.

### Project file & How to reproduce

Project file and screenshots included. To reproduce from scratch: create a model with a bridge or an overhang. Enable dynamic overhang speed in (Expert settings enabled) Print Settings -> Speed -> Dynamic Overhang Speed. Add a modifier to the model which contains the bridge or overhang, add a speed setting to that modifier which changes any of the dynamic overlap speed settings to something different from that set in Print Settings. Slice the model and change the color view to speed and note that the speed has not changed from the Print Settings speed.

Screenshot of model with several modifiers set to various different dynamic overhang speeds:
[image]
After slicing:
[image]

Project file (zipped .3mf) attached.
[bridge_test.zip](https://github.com/prusa3d/PrusaSlicer/files/11715361/bridge_test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta4-linux and 2.6.0-rc1-linux

### Operating system

Ubuntu 20.04

### Printer model

Ender 3 V2, SKR Mini e3 v3 mainboard, Klipper

Jan-Soustruznik, 2023-06-12:
SPE-1769

AssiveMass, 2024-04-08:
Until this is fixed, there should be a notification if dynamic overhang settings (speed and/or fan) and modifiers are on at the same time. Would save a lot of trouble.

Kekht, 2024-06-01:
Actually even in 2.7.4 even with overhang speed settings disabled, 
modifiers for bridge speed does not affect anything.

[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]