prusa3d/PrusaSlicer #10788: Sliced model does not obey boundaries between model and dynamic support

k8stuff, 2023-06-11:
### Description of the bug

When printing the attached Cheetah with dynamic support, it shows fine in the slicer, i.e. no parts of the model touch the support on the build plate - when printing however, the model overlaps with the dynamic support, causing the hot end to hit the print as the overlapping support / model area is now higher than the model.

Slicer view shows no overlapping

[image]

Print does show overlapping

[image]

Print does show overlapping - different view.

[image]

### Project file & How to reproduce

I am unable to upload project as size is not allowed (25MB allowed but archive is 29MB)

See zip on dropbox here : https://www.dropbox.com/s/d1nebt4clut6kja/DynamicSupportIssues.zip?dl=0
See rar on dropbox here : https://www.dropbox.com/s/agc6z3gp0utdoaa/DynamicSupportIssues.rar?dl=0

To reproduce simply slice and print. My mk4 is on order so unfortunately I can only try this on my Ender.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-rc1+win64-202306091137_signed

### Operating system

Windows 11

### Printer model

Ender 3 S1 Pro

k8stuff, 2023-06-11:
Sounds like same as #10766 although I only have a single model. Of course as it is ‘feet’ of an animal, they do look like individual models.

Jan-Soustruznik, 2023-06-12:
Hello, @k8stuff , 

could you share gcode, please?  thank you

k8stuff, 2023-06-12:
@Jan-Soustruznik 

[African_Cheetah_No_Base_1d0h4m_0.08mm_205C_PLA_ENDER3S1PRO.zip](https://github.com/prusa3d/PrusaSlicer/files/11726991/African_Cheetah_No_Base_1d0h4m_0.08mm_205C_PLA_ENDER3S1PRO.zip)

foreachthing, 2023-06-12:
I don't see an overlap in your gcode?!
AVI: https://drive.google.com/file/d/1SLzwUi4ojkQF387uNI92zTkcFVV9GjO7/view?usp=sharing

Your first layer:
[image]
Is not what's on your photo above... or I don't see it.

k8stuff, 2023-06-13:
@foreachthing 

If you read my original post you can see that slicer gcode appears to be fine but check the actual print fotos - they overlap. There’s a second issue showing the same problem.

neophyl, 2023-06-13:
If the gcode is correct then surely this is a printer problem ?

Jan-Soustruznik, 2023-06-13:
Hello, @k8stuff , We haven't this problem, pictures below are from mk4, I think that your first layer is too pressed.. or did you slice for the correct nozzel diameter? Becasuse it seems that you have too much material on pictures.

anyway, could you share more info about "I am not the only one that has reported this btw." I think that I dont know about this information, so If you are not only one whom have this problem, we should check it.

[image]
[image]

k8stuff, 2023-06-13:
Thanks @Jan-Soustruznik 

Glad it works on the mk4 so I just need to wait for it to be delivered :) 

As for 'I am not the only one' - I did mention the other issue in the comments. 

https://github.com/prusa3d/PrusaSlicer/issues/10766

Seems similar. I don't have multiple objects but the legs of the model would count as such I think

closed 2023-06-13 by k8stuff

sysdigmike, 2023-06-13:
Ok, thanks for the info ... Will try again when my mk4 is here.

foreachthing, 2023-06-13:
@k8stuff you could increase "XY separation between an object and its support" to avoid overlap.
https://help.prusa3d.com/article/support-material_1698#xy-separation-between-an-object-and-its-support

I read your post, but the images from your first post are not the same as the gcode you uploaded. In the gcode you did upload, there was no overlapping ... just saying.

k8stuff, 2023-06-13:
@foreachthing That was my problem / point though - the Slicer shows it correctly - no overlap, but the printer prints WITH the overlap and I am not sure we can increase the separation as I don't think there's much room between support and the model. 

So yes, the pictures are from the same gcode .. I exported the code, did another test print and got the same result. 

I would try the gcode with another slicer, but obviously Cura and the sorts don't do the organic support