prusa3d/PrusaSlicer #10790: more connector options for the cut tool

MacMadness, 2023-06-12:
Hi, i'm making figurines in blender and i always used connectors that are squared and narrow down at the top (look at example picture),
is it possible to add this kind of connector to prusaslicer instead of having a regular cube/cylinder? 
[image]

Jan-Soustruznik, 2023-06-12:
Hello, @MacMadness , 

Thank you for you feedback, I hope that I understood .. I think that this connectors are already implemented.

[image]

Jan-Soustruznik, 2023-07-14:
I closed issue as solved.

closed 2023-07-14 by Jan-Soustruznik