prusa3d/PrusaSlicer #10791: SL1 automatic arrangement does not work properly

Ronie80, 2023-06-12:
### Description of the bug

The automatic arrangement in Slicer on the SL1(s) does not work correctly. Although sufficient space is available, components are placed outside the print bed. The same components are placed correctly on MK3s and Mini.

### Project file & How to reproduce

Videofile:
https://github.com/prusa3d/PrusaSlicer/assets/78355022/114392b6-beeb-4981-af1b-035fb6d39409

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-rc1

### Operating system

Windows 11

### Printer model

Prusa SL1S

Jan-Soustruznik, 2023-06-12:
SPE-1754

FidelCapo, 2023-06-20:
Thank you for report. Should be fixed in the next release. Closing.

closed 2023-06-20 by FidelCapo