prusa3d/PrusaSlicer #10792: Negative Volume

kost111233, 2023-06-12:
### Description of the bug

Hi. I really like the built-in PrusaSlicer editor, but unfortunately not everything works as I want. I wanted to make a cutout from one part with the help of another, using the "Negative volume" function. The function works correctly, it cuts everything, but for some reason I can't save the cut part, it gives an error. There were no such problems on other parts. Does this function have limitations?

### Project file & How to reproduce

[image]
[negative_volume.zip](https://github.com/prusa3d/PrusaSlicer/files/11721626/negative_volume.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 beta 4.

### Operating system

windows 10

### Printer model

kingroon kp3s

sarusani, 2023-06-12:
You can't export it as STL in most cases. If you want to have the STL, you'll need to use a proper CAD editor.

(https://www.tinkercad.com/ is free and it's super easy if you just need same basic operations like negative volume)

kost111233, 2023-06-23:
> You can't export it as STL in most cases. If you want to have the STL, you'll need to use a proper CAD editor.
> 
> (https://www.tinkercad.com/ is free and it's super easy if you just need some basic operations like negative volume)

tinkercad is suitable for simple models, but for me it is very inconvenient and limited, models over 25 MB cannot be loaded into it. Prusaslicer is a great editor and I would like you to add this feature to the update, it will solve a lot of problems

foreachthing, 2023-06-23:
Duplicate issue like https://github.com/prusa3d/PrusaSlicer/issues/10818, and more.

Jan-Soustruznik, 2024-09-10:
closed issue - https://github.com/prusa3d/PrusaSlicer/issues/13169