prusa3d/PrusaSlicer #10795: Top Layer Height

yahbluez, 2023-06-12:
I think it would be useful to be able to set the "Top Layer Height" the same way we can set the "First Layer Height"

While there are several requests for more settings, 
it may be a good idea to have a switch more to enable more features:

simple - advanced - expert - **detail**

kubispe1, 2023-06-12:
Hi, what are some use cases for having the 'Top Layer Height' parameter available? Could you provide me with some inspiration, please?

rtyr, 2023-06-13:
One print job and even one object can have many top layers on different layers. Now imagine you are printing the whole layer of the big object with different layer height just because one tiny top layer area.

yahbluez, 2023-06-15:
> Hi, what are some use cases for having the 'Top Layer Height' parameter available? Could you provide me with some inspiration, please?

The use cases are the same why we have a separate setting for the first layer and why we have a setting for the extrusion wide of the top layer. Also we can iron top layers. Having the possibility to print the toplayer smoother is an advantage. 

Taking more care for the top layer gives us better top layers.
Other slicers already give the possibility to reduce the number of perimeters for the top layer to 1 to improve the visual quality.

yahbluez, 2023-06-15:
> One print job and even one object can have many top layers on different layers. Now imagine you are printing the whole layer of the big object with different layer height just because one tiny top layer area.

Good question! 
This is not a problem because we already can handle different toplayers individual. For example while ironing or print combined infill.

You can print with a layer height of 0.15 while doing the infill with 0.3mm.

The slicer knows what a top layer is and what not.

So the problem you imagine is not real.

rtyr, 2023-06-16:
> This is not a problem because we already can handle different toplayers individual.
> So the problem you imagine is not real.

Top layers are not using separate perimeters. Just slice simple stairs and look at layer preview at some top layer level.
 
> For example while ironing or print combined infill.

Not relevant to this issue.

yahbluez, 2023-08-07:
> Top layers are not using separate perimeters. Just slice simple stairs and look at layer preview at some top layer level.

A top layer is an infill not a perimeter. A toplayer may  have perimeters.

We already can change the extrusion wide of a top layer.

This feature request is about to get a setting allowing to set the layer height of a top layer to a seperate value, like we can do with the first layer.

Your argument on jun, 13 was that their may be more than one top layer and that may lead to a problem.
 I told you that the slicer knows what a top layer is because we already use this difference while ironing.

How can you then say that this is not relevant? It is relevant because it shows that there is no problem with handling top layer different because we already do.

neophyl, 2023-08-07:
With the example of steps given by rtyr, some of the layer is topfill and some of the layer is not, it could be sparse or solid infill, as well as the perimeters obviously.  The entire layer of ANY layer with even a tiny bit of topfill would need to be printed at your proposed layer height setting. Not just the small section that the ironing algorithm picks out and irons as a post process. 

There is only ever one first layer and it applies to the whole layer, no matter the feature type, there could be hundreds of layers that contain top layers, that’s the difference.  You can’t print different areas of a layer at different layer heights. If you want to do this then you can always just add layer height modifiers for the layers that you want to be thinner.

closed 2026-09-29 by prusaslicer-bot[bot]