prusa3d/PrusaSlicer #10796: Mouse ears instead of brim for larger objects

shledge, 2023-06-12:
Similar to how Cura does it - having a full brim on a large object seems to be a bit of a waste, when in reality it's more the corners that need to have extra grip.

I'm aware you can do it manually, but it's a rather fiddly and length process.

shledge, 2023-06-12:
Alternatively, paint on brims could be a good feature :)

neophyl, 2023-06-13:
Duplicate of many existing issues.

conkerts, 2023-11-06:
> Similar to how Cura does it - having a full brim on a large object seems to be a bit of a waste, when in reality it's more the corners that need to have extra grip.
> 
> I'm aware you can do it manually, but it's a rather fiddly and lengthy process.

How do you do it manually ?

https://www.youtube.com/watch?v=MCcFMDv_4eo
I came across this video and found it interesting idea. The sprue version sounds even better and pretty sensible to me.
Tho, unlike in the video, i think it should be a slicer option and not part of the object (or at best, an optional part of the object)

I dislike removing the brims, so brims for only the corners would be great. 

> Duplicate of many existing issues.
Is it at least on the radar of the devs ?

l29ah, 2024-01-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2954

Snuff1eupagus, 2024-04-26:
The Duplicate police won't help you, but this may. 

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

Good Luck.

closed 2026-09-29 by prusaslicer-bot[bot]