prusa3d/PrusaSlicer #10797: App crash when doing select all while in paint-on-supports mode

PaulN8, 2023-06-12:
### Description of the bug

App crashes when selecting a model then pressing paint-on-supports then going to the edit menu and choosing select all. I actually did that by accident and discovered this repeatable bug. I can't see how such an option would make sense anyway, perhaps select all should be greyed out in this case?
[image]

### Project file & How to reproduce

1. Load project
2. Click on either object
3. Press the paint-on-support button
4. Go to select all in the edit menu
5. Crash
[Box with clips.zip](https://github.com/prusa3d/PrusaSlicer/files/11728872/Box.with.clips.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-rc1-win64

### Operating system

Windows 10 64-bit

### Printer model

MINI+

Jan-Soustruznik, 2023-06-13:
Hello, @PaulN8 , Thank you for report. We will try to fix in next release. Have a nice day. 

SPE-1771

lukasmatena, 2023-06-14:
Should be fixed in [2.6.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc2). Closing.

closed 2023-06-14 by lukasmatena