prusa3d/PrusaSlicer #10798: Bug with Brim (Inner Only and Outer Only)

jimmysdiy, 2023-06-13:
### Description of the bug

I am using Prusa Slicer 2.6.0-rc1 on Windows 10 - 64bit OS 22H2, build 19045.2965

When using the following functions, 
1)  Inner Brim only -  after slicing the file, slicer does not generate inner brim. 
2) Outer Brim Only - when slicing the file, slicer generates inner and out brim together despite selecting outer brim only

### Project file & How to reproduce

[Grommet Slotted.zip](https://github.com/prusa3d/PrusaSlicer/files/11729445/Grommet.Slotted.zip)

1) [image]
1) [image]
 
2) [image]
2) [image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Pursa Slicer 2.6.0-rc1

### Operating system

Windows 10 - 64bit OS 22H2, build 19045.2965

### Printer model

Ender 3 Pro

neophyl, 2023-06-13:
Your object only has an outer perimeter.  Due to the slot the perimeter can be traced completely around your object.  As such it’s all outer. So no bug.

jimmysdiy, 2023-06-18:
That is incorrect. It works fine in Cura. Secondly, Brim shouldn't happen inside the model. 

The Slicer has an option for Inner, Outer and Both. 

What is inner good for? Why would you need inner brim if the bottom is touching the plate anyways?

On Linux, it works fine. I've verified with some folks.

neophyl, 2023-06-18:
Im not sure who you have verified it with but its simple geometry.  Your outer perimeter -
[image]
Note you can trace its path without break.  That means its s single OUTER perimeter.  

As far as brim generation is concerned Inner is used where there is something like a hole.  For example this simple cylinder has an inner and outer perimeter.
[image]
Note that it has 2 sections of orange perimeter, that can not be traced continuously and that one is inside the other.  Also note that brim in only generated on the outside, that's with the settings from your project.  

Both objects together clearly showing that the external brim only function is working correctly -
[image]

closed 2026-09-29 by prusaslicer-bot[bot]