prusa3d/PrusaSlicer #10802: Invalid spacing supplied to Flow::with_spacing()

guestisp, 2023-06-13:
### Description of the bug

When slicing `CUB rudder 3.stl` from the Piper J-3 Cub by 3D LabPrint, the error `Invalid spacing supplied to Flow::with_spacing()` is triggered.

I've attached the official 3mf file as this project is free
[CUB rudder 3.zip](https://github.com/prusa3d/PrusaSlicer/files/11737267/CUB.rudder.3.zip)

### Project file & How to reproduce

Just slice the attached 3mf to trigger the error

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-rc1+linux+x64-GTK3

### Operating system

Ubuntu 23.04

### Printer model

BLV mgn Cube

Jan-Soustruznik, 2023-06-14:
Hello, @guestisp , 

I checked your 3mf file, there is invalid value in the Bridge flow ratio = "80" / max valid value = "2" ... Could you help me where did you find this profile? Or how did you input this value? Thank you

guestisp, 2023-06-14:
here: https://3dlabprint.com/shop/piperj3cub/

(a better error message would be aprecieted)

Jan-Soustruznik, 2023-06-15:
@guestisp , this protection is already implemented. You cannot input value bigger then 2 for Bridge flow ratio ... so main point for solvet this issue is understood how you did value bigger than 2, then we can add additional or modify protection... Could you try to remember how did set this value in your project, please? 

Sorry I cannot find any info about the Bridge flow ratio = "80" or some test 3mf where is set this value.

> here: https://3dlabprint.com/shop/piperj3cub/
> 
> (a better error message would be aprecieted)

neophyl, 2023-06-15:
Jan in the past when using older versions of both PS and Super Slicer and opening 3mf projects that were created in the other I have had issues where values that were 0.9 for example changed to 90.  Both ways Prusa<>SuSi.  I think I have seen that in the past with bridge flow ratio too.

So while you may have protections around manually editing the field  I suspect further checks will be needed around the opening of 3mf files.

guestisp, 2023-06-15:
@Jan-Soustruznik as wrote, the 3mf isn't mine, it's a 3mf that you can purchase from 3d labprint. This particular case, it's a free 3mf (that i've already attached in this issue). So, why are you looking where to get the 3mf if the same 3mf is already attached ? Probably they made the 3mf with an older version of prusaslicer. it should be wrote in 3mf, right ?

Anyway, is the error is due to a wrong value for bridge flow ratio, it should be wrote in the error message to....

closed 2026-09-29 by prusaslicer-bot[bot]