prusa3d/PrusaSlicer #10809: configuration update, template 1.0.3 Update compatible Printer ....

j-snijder, 2023-06-14:
### Description of the bug

after loading this configuration the bed moves way down and crashes on the stepper motors when you slice and print.

### Project file & How to reproduce

Sorry, I totally paniced so forgot to take time to add project files and so on.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 rc1

### Operating system

windows 11

### Printer model

Zonestar Z9V5pro

j-snijder, 2023-06-14:
Then why, after trying for a few times, it does happen ???

Is there a built-in preset update done today ?

rtyr, 2023-06-14:
> Then why, after trying for a few times, it does happen ???

I don't know. I am afraid we can't troubleshoot issues with your printer. As I said, the "Template" bundle contains only filament profiles (and the change in 1.0.3 is related only to Prusa printers).

Only thing I can do for you is to check your project file if I can see any issue in your start g-code for example.

> Is there a built-in preset update done today ?

No, the last Zonestar update is roughly 5 months old.

rtyr, 2023-06-14:
These sort of issues can also happen with corrupted filesystem on sd-card/usb drive. You may try to format it.

j-snijder, 2023-06-14:
strange, when these things happen, and not once but a few times as I checked it against config snapshots from yesterday and today :)
Okay, I've reported it, see if you can find the solution, if not its too bad .

I have rolled back to a config from yesterday evening end that works without the problem, so I am okay now

rtyr, 2023-06-14:
As I said. If you provide problematic project file or at least the g-code we can check if it is ok. I will close this issue in other case.

j-snijder, 2023-06-14:
Can close the issue

Verzonden vanuit Mail voor Windows

Van: rtyr
Verzonden: woensdag 14 juni 2023 16:01
Aan: prusa3d/PrusaSlicer
CC: j-snijder; Author
Onderwerp: Re: [prusa3d/PrusaSlicer] configuration update, template 1.0.3 Updatecompatible Printer .... (Issue #10809)

As I said. If you provide problematic project file or at least the g-code we can check if it is ok. I will close this issue in other case.
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

j-snijder, 2023-06-14:
Can Close

closed 2023-06-14 by rtyr