prusa3d/PrusaSlicer #10813: Floating Bridge Anchors

ForceConstant, 2023-06-14:
### Description of the bug

I get warning about Floating Bridge Anchors, but have auto bridge angles on. 

[image]

### Project file & How to reproduce

[calicat_calico.zip](https://github.com/prusa3d/PrusaSlicer/files/11749574/calicat_calico.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 rc1

### Operating system

Windows 11

### Printer model

ender3

ddacunha, 2023-09-12:
This is where it happens, we can see that the Bridge infill is not properly anchored. Looks related to this bug #9999 
[image]

ddacunha, 2023-09-12:
My bad, I believe it happens at a lower layer:
[image]

natecan, 2024-02-02:
I have found the solution, set your bridging angle to anything above 0 (I have set mine to .00000001) and it will properly anchor to objects.  I believe that the issue is attributed to the use of a zero
[image]
[image]
.

closed 2026-09-29 by prusaslicer-bot[bot]