prusa3d/PrusaSlicer #10814: 2.6 should not open always in full screen

Mundball, 2023-06-14:
### Description of the bug

since Prusa Slicer 2.6, always open in full screen.

I test the 2.6 version first time with beta 2 linux-x64-GTK3

### Project file & How to reproduce

1. Open Prusa Slicer
2. now is full screen

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-rc2+linux-x64-GTK3

### Operating system

Debian 11

### Printer model

Prusa i3 mk3 + mmu2

blester125, 2024-07-05:
I agree, I also see this behavior in 2.8 on ubuntu 22.04. I can't seem to find a way to disable starting in full screen and it's really annoying to have to turn it off each time

leftyfb, 2025-01-02:
This still happens on 2.9.0 and is very annoying.

closed 2026-09-29 by prusaslicer-bot[bot]