prusa3d/PrusaSlicer #10816: Dynamic overhang speeds and fan settings not applied correctly in vase mode

awabom, 2023-06-15:
### Description of the bug

When in spiral vase mode, the dynamic overhang speed and dynamic fan speeds are only applied to the 'underside' of a sloping model. The top surface is not treated as overhanging - even though it is, since we are in vase mode?

### Project file & How to reproduce

[dynamic_overhang_vase.zip](https://github.com/prusa3d/PrusaSlicer/files/11756691/dynamic_overhang_vase.zip)

Slice and visualize 'fan' or 'speed' and notice that only the 'underside' surface gets dynamic speed/fan applied.
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-rc1+win64

### Operating system

Windows 11

### Printer model

Ender 3 V2 (modded)

awabom, 2023-06-15:
This still applies to 2.6.0-rc2 (#10786 sounded similar?)

Jan-Soustruznik, 2023-06-16:
Hello, @awabom , 

Thank you for your feedback. Currently dont check overhangs for the "internal" perimeters. Because there is mostly some infill. We will modify your issue to feature request. 

btw, #10786 is different thing - this is issue, that we will try to solve.

Have a nice day 

thank you, have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]