prusa3d/PrusaSlicer #10819: Missing Modifiers

bentdragon, 2023-06-15:
### Description of the bug

I create a project with modifiers and negative part, all looks good and when I slice it, the add on parts are missing? Go back to standard view and they are still there. As it is a 4 hour print, I don't want to waste the filament to try. Tried three different projects, same results.

### Project file & How to reproduce

[Flat_funnel_remod_100_mm.zip](https://github.com/prusa3d/PrusaSlicer/files/11761075/Flat_funnel_remod_100_mm.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

ver 2.6.0-rc1+win64

### Operating system

windows 10

### Printer model

Mk3+

neophyl, 2023-06-15:
The problem is that it **is** a modifier.  Your added on block is type modifier.  That means it modifies the settings of objects/parts.   It isn't a part and so doesn't generate perimeters or infill of its own.   You have to change its type to Part.  Right click and change type.

[image]

Here is a modified project as an example.  I've also removed all the unnecessary support to show it clearer.  
[Flat_funnel_remod_100_mm_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/11763586/Flat_funnel_remod_100_mm_neo.zip)

Jan-Soustruznik, 2023-06-16:
Hello, @bentdragon , 
as @neophyl mentioned above, modifiers doesn't generate volumes, by modifiers you can update local print settings, see our knowledgebase for more info: https://help.prusa3d.com/article/modifiers_1767

Anyway your used model is corrupted - there is lot of open edges and missing layers. If you dont want to waste a material, you should consider repair model before print, more info you can find also on our knowledgebase base: https://help.prusa3d.com/article/corrupted-3d-models-for-printing_2205

I closed issue as solved. 

thank you, have a nice day

closed 2023-06-16 by Jan-Soustruznik