prusa3d/PrusaSlicer #10820: error M106> 255

Serj121, 2023-06-15:
### Description of the bug

when forming the g code, the fan speed is greater than 255
M106 S280.5

### Project file & How to reproduce

[Big button 1.75_0.08mm_ABS_MK3S_37m.zip](https://github.com/prusa3d/PrusaSlicer/files/11764041/Big.button.1.75_0.08mm_ABS_MK3S_37m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Версия 2.6.0-rc2+win64

### Operating system

windows 11

### Printer model

kingrun kp3s pro

lukasmatena, 2023-06-16:
@Serj121 Thanks for the report. I unzipped your 3MF and I noticed that configuration value `bridge_fan_speed` is set to 110 %. May I ask how you created the 3MF? PrusaSlicer should not have let you to set the option above its maximum (which is 100 %).

Serj121, 2023-06-16:
[image]

Serj121, 2023-06-16:
[PrusaSlicer profiles.zip](https://github.com/prusa3d/PrusaSlicer/files/11772678/PrusaSlicer.profiles.zip)

closed 2026-09-29 by prusaslicer-bot[bot]