prusa3d/PrusaSlicer #10821: Dynamic extra length on layer restart

robustini, 2023-06-16:
**Is your feature request related to a problem? Please describe.**
Printing airfoils with a very low infill and with the extrusion flow calibrated micrometrically to have the minimum weight and a good resistance I've this big problem.
As you can see from the image at the layer change when it starts to extrude it lacks material, but it does not always do so, only when the travel of the nozzle is long enough.
At that moment the nozzle empties itself, necessarily printing PLA at about 230°, which is imperative temperature this type of printing.
So by repeating the print the result is the same, always missing in the same points.
It does not happen where the travel of the nozzle is low, that is when it has to make little movement to restart with the new layer.
Unfortunately the"Same position: Aligned" works relatively, often the nozzle travels from the leading edge to the trailing edge of the wing at the end of the layer, thus creating this problem.
It could be solved by inserting an "extra length on restart" value, but this would also affect where the problem does not arise, thus creating an accumulation of material where now starts well.
It would therefore be interesting to have a "dynamic extra length", calculated during the slicing over a nozzle travel distance, and which is obviously modular.

**Describe the solution you'd like**
Example: if I set "dynamic extra lenght" to 1mm on 100mm of travel, if the movement before the restart of the nozzle will be 50mm it will have an extra length of 0.5mm, therefore the value should dynamically adapt according to the length of the travel.
This would permanently solve this problem, without creating artifacts where it is not needed.
I don't think there is this feature in any slicing software, so it would be very nice if Prusa Slicer had it.

**Additional context**
A result without the extra lenght on restart is visible here.
[image]

Here, however, we exaggerate that parameter.
You can see that when the travel is high the result is great, when the travel is short it creates a buildup on the seam.
[image]

I made the same request on the SuperSlicer repository, which I link to for reference.
https://github.com/supermerill/SuperSlicer/issues/1032

robustini, 2023-06-16:
Some might comment that my printers are set wrong.
I post a few pictures to make it clear that they print perfectly, try to locate the seams in these photos, you can't see them.
The feature I am asking about is related to that kind of printing with temperatures higher than what the filament manufacturer states.

[image]

[image]

neophyl, 2023-06-16:
Your wording is different but I think this is essentially the same request as https://github.com/prusa3d/PrusaSlicer/issues/10311  (trying to fix the same problem anyway).

robustini, 2023-06-16:
> Your wording is different but I think this is essentially the same request as #10311 (trying to fix the same problem anyway).

I have read and conceptually the proposed solution is very similar, although I think what I propose is simpler in terms of implementation.

closed 2026-09-29 by prusaslicer-bot[bot]