prusa3d/PrusaSlicer #10822: add 'hole horizontal expansion' feature

goddengo, 2023-06-16:
**Is your feature request related to a problem? Please describe.**
when print module with hole, the module from my print has accurate outline size, but hole size smaller than expectation. 
to calibrate it, if I adjust "XY_size_compensation" (eg. set to -0.14), hole diameter become good, but meanwhile diameter of outline become smaller. 
seems no individual configure to compensate hole diameter, such as 'hole horizontal expansion' in CURA. 
it will be much perfect if PrusaSlicer can introduce the feature in.

**Describe the solution you'd like**
it will be much perfect if PrusaSlicer can introduce the feature in.

**Describe how it would work**
compensate hole diameter separately, not impact other size.

**Describe alternatives you've considered**
N/A

**Additional context**
'hole horizontal expansion' feature in ultimaker CURA can be referred

neophyl, 2023-06-16:
Duplicate of so many issues.  Earliest being https://github.com/prusa3d/PrusaSlicer/issues/515 I think.

Mekannick, 2023-06-27:
XY hole compensation WHEN???

Googliola, 2023-07-06:
Prusa, I really love your slicer and all the fancy new features. But hole-compensation is an absolute must-have. Totally inexplicable why it has not been implemented yet. 
Don't tell your users to account for that in design, as it is a) plain wrong approach and b) not feasible when you download STL that did not account for it. 
### Version 2.6.1 ??? Pretty please!

RandomGgames, 2023-08-25:
I'm surprised Prusa Slicer doesn't have separate XY compensation for holes yet. It's one of the few reasons, while I love the slicer, I don't use it as my main and I've been experimenting with a couple other free slicers/forks like superslicer (which I love so much but it's not being updated anymore it seems) and orcaslicer. Cura is currently my go to.

MapleDodoKiwi, 2025-03-10:
Anyone know if there has been any progress on creating this feature from Prusa?

closed 2026-09-29 by prusaslicer-bot[bot]