prusa3d/PrusaSlicer #10825: Prusa Slicer online manual page for 'insert pause at layer' is broken

matmire-makes, 2023-06-16:
### Description of the bug

Prusa Slicer online manual page for 'insert pause at layer' is broken.  Linked page is showing error 500.  It has been like this for a while.  https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490#insert-pause-at-layer

### Project file & How to reproduce

go to this link... https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490#insert-pause-at-layer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

na

### Operating system

na

### Printer model

na

kubispe1, 2023-06-18:
Looks like that is already working.

closed 2023-06-18 by kubispe1

matmire-makes, 2023-06-29:
It's down again.  This seems to happen a lot.  It works one day, then stops working error 500 and 404 page.  I link my users to this page for a printing guide, and its not working way too often.