prusa3d/PrusaSlicer #10826: empty Space Error connecting to Supports on 2.6 Alpha, Beta, RC2

evanquinnalaska, 2023-06-16:
### Description of the bug

Slicer generates empty spaces between supports and STL.

### Project file & How to reproduce

Delete c:\users\username\appdata\Roaming\Prusa Folders
Fresh Profile > mk3s .4 nozzle > Next to finish.
Print Settings .10 DETAIL
Import file > Supports > Support on Build Plate only > Print Settings > Supports > Organic > Slice. 
[image]
[image]
[image]
[image]
Cannot upload Project file (113mb file size)
Also attempted a file repair to see if that would fix it same results. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 RC2, Beta 3, Beta 6, Alpha

### Operating system

Windows 10 (19044.3086)

### Printer model

mk3 s

sarusani, 2023-06-17:
Looks like the details are too small to slice with your arachne settings.
Try to adjust them (print settings->advanced->arachne perimeter generator) or select the classic perimeter generator (print settings->layers and perimeters->perimeter generator).

Not sure, but I think that should fix it.

evanquinnalaska, 2023-06-19:
Sarusani you are correct, and your work around fixed it.

closed 2026-09-29 by prusaslicer-bot[bot]