prusa3d/PrusaSlicer #10827: Unwanted solid infill added to perimeters

mpaw, 2023-06-16:
### Description of the bug

I have an object that has 5 bottom layers, 0 top layers, 3 perimeters, and 0% infill. PrusaSlicer keeps adding unwanted solid infill next to the inside perimeters. I have tried every setting possible and I can't find a way to disable this

[image]

### Project file & How to reproduce

[Solid_Infill_Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/11776465/Solid_Infill_Issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-rc2+MacOS-arm64

### Operating system

macOS 13.2.1

### Printer model

Ender 3 Pro

mpaw, 2023-06-28:
Can someone take a look at this issue? It's extremely annoying that there is no way to turn this off. Below is a screenshot from SuperSlicer using the same settings and it does not happen.

[image]

neophyl, 2023-06-28:
Super Slicer has extra settings to stop it happening that are configured by default, that for many objects but are configurable to adjust how it works.  So the settings are not the same.

mpaw, 2023-06-28:
@neophyl I meant the settings I specified in my first post.

> I have an object that has 5 bottom layers, 0 top layers, 3 perimeters, and 0% infill

Obviously SuperSlicer has more settings than PrusaSlicer. Regardless, I doubt what PrusaSlicer does in this case is behavior that anyone actually wants. You get disjointed pieces of filament that provide no structural integrity.

neophyl, 2023-06-28:
This is a long running issue, a duplicate of so many, for example https://github.com/prusa3d/PrusaSlicer/issues/1054 
Recent changes in 2.6 has stopped the short zigzags of the solid infill but haven't done anything about turning it off completely.
I think the original issue is https://github.com/prusa3d/PrusaSlicer/issues/223 from 2017.

jmerc77, 2023-07-04:
having the same issue of the disable setting being removed. there used to be ensure vertical shell thickness, witch when unchecked, disables that purple lines in perimeters. now gone. a better solution than a check box would be an overhang threshold angle where if the part overhangs inward more than this angle, it enables. otherwise it defaults to off. if you set this angle to 90 (horizontal) it disables entirely.

CodingDutchman, 2023-07-14:
Please bring back the "Ensure vertical shell thickness" checkbox or give us another way to disable this. I have a hollow item that is illuminated from the inside and it would look horrible with this extra infill.

xTremePower, 2023-10-24:
Version 2.7.0 alpha .... and still not a single fix in Prusa ...
Problem fixed in SuperSlicer and also in bamboolab slicer .... But Prusa team continue to completly ingore this big waste problem since years, no matter how many reports we do ....

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena