prusa3d/PrusaSlicer #10829: Print Settings/Advanced/First layer message disagrees with value.

marhar, 2023-06-17:
### Description of the bug

The default value is 0.65, the same as the extrusion width.

The popup help states that the default is 200%.

at 0.65, bed adhesion is poor.  at 200%, bed adhesion is excellent.

Suggest changing the default value to that mentioned in the popup, it is a great improvement.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha

### Operating system

MacOS

### Printer model

Prusa Mk 3

rtyr, 2023-06-17:
200% is default value of PrusaSlicer as an application, it is not related to any particular printer.

As tooltip states, percentage is calculated over first layer height. So for example if first layer height is 0.2mm, then extrusion width would be 0.4mm.

closed 2023-06-17 by rtyr

marhar, 2023-06-17:
Hi, I think we misunderstand.

There is a bug.

The default value SHOULD BE:   200%

The default value IS:  0.65

Fix: **change the default value to 200%.**

Sorry for the misunderstanding!

rtyr, 2023-06-17:
No, 0.65 is not a bug. 200% would be bug (0.4mm extrusion width with 0.6 nozzle).

marhar, 2023-06-17:
It's not me saying 200%, it's the software saying 200% :)

My only point: Make the default and the message about the default not
disagree.

PS, thanks for all your work on this, I use PrusaSlicer every day and it is
wonderful software!

rtyr, 2023-06-18:
If you will use any sort of percentage, it will be calculated over first layer height. First layer height is set to 0.2mm in our 0.6mm nozzle profiles, therefore using 200% would be a mistake.